iliasam/Laser_tape_reverse_engineering #3: Looking for Laser ranger's alternative wiring/circuit layout.

pcbtborges, 2020-05-17:
Hi, I have acquired the module on the link below.
Can it be used?
Have you got the wiring diagram for this model?
Thanks in advance
Paulo

https://pt.aliexpress.com/item/4000410794896.html?spm=a2g0o.detail.1000014.1.120b73f7wVKQES&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.13338.170520.0&scm_id=1007.13338.170520.0&scm-url=1007.13338.170520.0&pvid=afaae825-5213-47db-b141-89255d4e7e8f&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.13338.170520.0,pvid:afaae825-5213-47db-b141-89255d4e7e8f,tpp_buckets:668%230%23169576%232_668%23808%235965%23504_668%23888%233325%2317_668%232846%238107%2351_668%232717%237558%23137

iliasam, 2020-05-17:
Hello.
If it is really 703A module, it could be used.
A had added pinout: https://github.com/iliasam/Laser_tape_reverse_engineering/blob/master/PCB_photos/module_703A_pinout.png

pcbtborges, 2020-05-17:
Hi, thanks.
How about the rest of the parameters, serial speed and board pin-out, is everything the same?
Also, can you please confirm VCC for that board? Online information is so erratic.
I acquired the module as is and not as a ready to use Range Finder, will there be a need to change firmware?
Thanks in advance
Paulo

iliasam, 2020-05-18:
Yes, all other parameters should be same.
703A module is designed for battery powering, so it need 3V for working. But 3.3V is suitable too.
  
The module by your link is declared to have a special firmware from the factory, that is supports communication with PC.

closed 2020-05-20 by iliasam