iliasam/Laser_tape_reverse_engineering #4: Accuracy and Range Limits

epsi1on, 2021-02-14:
Can you please describe how much is Range limit of measured distance?
The original laser meter is marked as 40m, but what about the new firmware?
Can it measure distances more than 40m by prism reflectors?

Thanks

iliasam, 2021-02-14:
Modulation frequencies are selected so that max range is 37.5m. If you need more, you have to change firmware.
I'm not sure that you will get enough signal at such range with my firmware - it was designed for high speed measurements.
Prism reflector or just reflective plate could probably help.

epsi1on, 2021-02-14:
Thanks for information. The phase difference method for calculating the distance is also used on equipment named `Total Station`. In the advertisement they say in each KM with their own special prism there is a few millimeters error.
[image]

What method do they use to first get a KM range, and also such good accuracy of mm?

Thanks heaps

iliasam, 2021-02-14:
As I know, total stations could use low frequency (1-10 MHz) modulation for initial (not accurate) big distance measurement.
After that, station is switched to high frequency mode to increase resolution.
Of course, total station is based on high complex electronics, and have a lot of calibrations to get high precision.

closed 2021-02-16 by iliasam