iliasam/Laser_tape_reverse_engineering #5: 85A module not working.

MickTheMechanic, 2021-11-07:
In have purchased the following device, which appears to be identical to the D30 device:
https://www.amazon.de/gp/product/B08H8WPB3M/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

Upon disassembly i noticed the module was marked as 85A, but it appears to be visually identical to the 85B module:
[image]
[image]

On the JRT-Measure website it seems like the 85A is the same as the 85B, just with a more powerful laser diode, better suited for industry applications:
https://www.jrt-measure.com/distance-sensor-short-range/57266349.html

I am able to succesfully program the MCu via the STLink software using the U85B HEX files, but I do not get any data via the serial connection, and the laser is permenantly on.

I have tried using an arduino and a CH340 shield to interface with my PC, and I have tried the serial monitor in the Arduino IDE, Serial Port Monitor, and the PC utilities, but I am not getting any comminication from the module.

Could you provide some support for troubleshooting?

iliasam, 2021-11-07:
What is the name of HEX file that you are using?
Do you use ST-Link programmer or DFU bootloader?

MickTheMechanic, 2021-11-07:
I have tried STLink programmer and STcube programmer.

I have tried the following files from the CortexM0 folder:
Firmware_dist_calculation_fast_boot.hex
Firmware_dist_calculation_fast.hex
Firmware_dist_calculation_simple.hex
Firmware_raw_capture.hex

Each upload appears to be sucsessful, but no communication via serial. 

It is possible i have made a mistake somewhere, it is my first time working with STM32 products, until now my experience is only with ATMega and ESP32 products.

MickTheMechanic, 2021-11-07:
I checked the documentation again, I think i have made a mistake.

When using the programmer which file do I select?

iliasam, 2021-11-07:
Try to use "Firmware_dist_calculation_fast_boot.hex" again with "PC_show_results.exe" utility.
Your USB-UART converter should support baudrate - 256000.
It would be good to check TX pin of the module by oscilloscope.
Also you can try to disconnect beeper from PCB and connect it between TX pin and GND. You should hear buzzing if module is transmitting data (I tried it).

If you are using NOT "_boot.hex" file, you need to connect module UART RX pin to external UART which will pull up this line - to get UART working.
"_boot.hex" files do not have such check , UART is always enabled in such firmware.

iliasam, 2021-11-07:
Sorry, my mistake BOOT0 must be connected to GND or it can be floating to get firmware running.

MickTheMechanic, 2021-11-07:
In the STCube software which baudrate should i select for the uplaud via UART?

iliasam, 2021-11-07:
Baudrate for bootloader is 115200 - you can see at image: https://github.com/iliasam/Laser_tape_reverse_engineering/wiki/U85-module-programming
I have updated text and pinout about BOOT0 pin just now, be careful.

MickTheMechanic, 2021-11-07:
I cannot connect, i have the error message: 

```

17:07:15 : Serial Port COM4 is successfully opened.
  17:07:15 : Port configuration: parity = even, baudrate = 115200, data-bit = 8,                     stop-bit = 1.0, flow-control = off
  17:07:17 : Timeout error occured while waiting for acknowledgement.
  17:07:17 : Timeout error occured while waiting for acknowledgement.
  17:07:17 : Error: Activating device: KO. Please, verify the boot mode configuration and check the serial port configuration. Reset your device then try again... 
  17:07:21 : RTS low
  17:07:21 : DTR low

```

I currently have PWRON and BOOT0 connected to VBAT, and NRST is floating. Is this correct?

iliasam, 2021-11-07:
"PWRON and BOOT0 connected to VBAT, and NRST is floating." - this is correct.
So it is strange, that you can't connect from bootloader.
Are you sure that RX and TX are connected OK (not swapped)?

MickTheMechanic, 2021-11-07:
I have tried both connections, but i cant get the connection to work. Is it possible i have damaged the bootloader somehow?

iliasam, 2021-11-07:
Strange situation.
st-link programmer with NRST connected and  "Connect under Reset"  selected should we working anyway.
What is the type of st-link programmer that you are using?

MickTheMechanic, 2021-11-07:
I can connect with the STlink, but not with serial. I am currently re-doing all my connections, incase it was an issue with the cables or soldering.

I have the STLink V2, looks just like a usb thumb drive

iliasam, 2021-11-07:
If your programmer is same: https://www.mikroshop.ch/shoppic/ST_LINKV2.jpg
You need to notice, that it do not have NRST pin. Its RST pin for STM8 only: https://research.kudelskisecurity.com/2020/04/15/swd-part-3-swo-and-nrst/
But if can erase and flash MCU without NRST connected to st-link, so that is OK.

MickTheMechanic, 2021-11-07:
It looks like i damaged the module somehow, I have a second one here and I have sucsessfully connected via Serial

MickTheMechanic, 2021-11-07:
I have done the calibration at >10cm, but the distance is way off. It is showing 2.8m at a distance of only 1.7m. 

Are there further steps that need to be taken to improve accuracy?

iliasam, 2021-11-07:
Could you try to go calibration at distance 20cm?
I do not have such big errors at my module.

MickTheMechanic, 2021-11-07:
Ok, i will do some testing and let you know how it goes. 

I noticed some components on the circuit board are different when compared to the U85B, most notibly is the large component (an inductor?) is no longer present, and the 6 pin IC marked FB3MKE has been replaced by 2 smaller 5 pin IC's labelled YJ33. Do you know what this circuit is responsible for? 

Also, do I understand correctly that the _simple file is for precision, _fast file is for speed? 

And a final question for now, i have been observing the signals in the PC utility, and I noticed the low frequency (FREQ LOW) signal does not look clean, whereas the other frequencies are a very clean sinus wave. Is this normal?

[image]
?

iliasam, 2021-11-07:
FB3MKE IC (DC-DC) is used for producing 3.3V for power supply. YJ33 is a linear source of 3.3V.
I'm afraid that "85a" version could really need other phase compensation method.

"Also, do I understand correctly that the _simple file is for precision, _fast file is for speed?"
No, they have same precision, but "_simple " is just written simpler to understand, but it is slower.

" low frequency (FREQ LOW) signal does not look clean"
That is OK, this "FREQ LOW" and "FREQ5" are experimental.

MickTheMechanic, 2021-11-07:
Do you mean temperature compensation? 

Another question I have is about the photosensor voltage, according to the pc utility it is at 110v, is this normal? And should this value change or remain constant?

iliasam, 2021-11-07:
Phase is compensated by measured amplitude value, temperature phase offset is compensated too, but temperature dependence is much smaller.

110v is OK, so be careful with module. Module does not produce really dangerous current, but such voltage hurts and could easily destroy low-voltage electronics if some short circuit happens.  
APD voltage is changed when measured temperature is changing.

MickTheMechanic, 2021-11-07:
I would like to get this module working correctly, it may also benefit others in the future.

could you walk me through how to determine a new phase compensation for this hardware? I can do testing or take measurements with an oscilloscope, I just don’t know where to begin.

iliasam, 2021-11-08:
I can try to give you description how to collect needed data.
Firstly, I need to build firmware with compensation disabled - it must be used with "PC_phase_analyse.exe" utility.
You would not need any other electronic devices co collect data.
But you need to have some kind of movable plate that must be installed close to laser output lens.
This plate is needed to change amplitude of transmitted light accurately.  See this photo: https://habrastorage.org/r/w1560/web/992/4ae/5f5/9924ae5f5807467fa40f9595134ac2f4.jpg

MickTheMechanic, 2021-11-08:
Ok no problem, I can manage some kind of mechanism for that, let me know when you have the modified firmware.

i have read your article in more detail and i have some observations and questions..

with the current firmware I observe that the distance measurement is not correct, one mm measured from the sensor corresponds to multiple centimeters, apart from that the distance seems stable until a distance of about 30cm (95mm according to the sensor) is reached, then the value jumps up to over 1000mm, it is the stable once more for a short distance before jumping again.

I wonder if this phenomenon is caused by the phase difference exceeding 360°, Or perhaps the reference frequency is no longer 5khz with the new hardware? Just some thoughts I had while reading your paper.

also thank you for the support 👍

iliasam, 2021-11-08:
All frequencies are generated by PLL IC (si5351 or its clone), which is same type in all types of modules. This IC is fully controlled by  MCU, so all frequencies should be OK.

MickTheMechanic, 2021-11-08:
Ok, thank you for the explanation.

I will begin work on the mechanism for limiting the Laser output.

iliasam, 2021-11-08:
I have published test firmware: https://github.com/iliasam/Laser_tape_reverse_engineering/blob/master/Code/CortexM0/test_firmware_phase_calc.hex
See readme.txt description: https://github.com/iliasam/Laser_tape_reverse_engineering/tree/master/Code/CortexM0
I need some time to write description how to make measurements.

MickTheMechanic, 2021-11-08:
Thanks, I have also finished the device for limiting the amplitude:
[image]
[image]
[image]

I await your further instructions.

thanks for the help

MickTheMechanic, 2021-11-08:
In your reply you said the plate should cover the laser output, but in the image it seems like the plate is covering the photo sensor input. Should the input or the output be covered?

iliasam, 2021-11-09:
Yes, in my photo plate is closing photodiode lens, but photo is made with old module. I had closed laser output with 85b module.
I suppose that your device should work OK. You can test it - load test firmware, run  "PC_phase_analyse.exe" utility and check amplitude at different positions of plate.

MickTheMechanic, 2021-11-09:
I have tested the device and I can adjust the amplutude by adjusting how much the laser is covered.

What are the next steps?

iliasam, 2021-11-10:
Needed measurement steps:
1. You need to fix module somehow so that its beam should not move when you are changing plate position.
2. You need to point laser beam to some white object, this object should be movable is range 0-1.5m from module.
Movement direction should be parallel to laser beam (you don't need super high accuracy here).
3. Power on module, open photodiode plate, and wait 5 min to get warm up. Run "PC_phase_analyse.exe" and confirm that values are measured correctly.
You need to control values:  "Amplitude" and "Freq 1 Phase AVR". "AVR" is averaged value, you can see that this value is changing slowly, so you need to wait for a certain time until this value will get stable after any changing of plate position.
4. Move object along laser beam until "Freq1 phase" will be max close to 0.0 deg, align object position to get "Freq 1 Phase AVR" closet to 0.0 deg too. This would be **zero distance position**.
5. Change position of photodiode plate by small steps so amplitude should decrease to ~100 steps.
It would be good to keep amplitude close to x00 values.
So you need to create table like:
Amplitude1000 - phase#1
Amplitude900 - phase#2
...
Amplitude100 - phase#A
Amplitude50 - phase#B
6. You need to move object to 230 mm from **zero distance position** (check distance by ruler) and do same measurements as in step 5.
7. You need to move object to 460 mm from **zero distance position** (check distance by ruler) and do same measurements as in step 5.
Share measured 3 tables.

MickTheMechanic, 2021-11-11:
[AmplitudeValues.ods](https://github.com/iliasam/Laser_tape_reverse_engineering/files/7523480/AmplitudeValues.ods)

MickTheMechanic, 2021-11-11:
Is this what you need?

iliasam, 2021-11-12:
Thanks for data. 
All measurements are made correctly.
Sorry, but I need to say that this data are hard to analyze because it looks like you get signal overdrive.
It would be great, if you can repeat measurements but do small a change in method.
You need to change object (use darker object) so that amplitude value should never be never bigger than 1600.
I suppose that you will get another zero distance position in this way.
Also, zero distance position should be more than 30cm from the module.
You can do increase amplitude steps to 200 to save your time, but you still need to make measurements at 50 and 100 amplitude values.

MickTheMechanic, 2021-11-15:
I have re-measured the values. I changed the surface from a gloss white to a matt white, and made a new 0 point whaich was 500mm from the sensor.

Here are the new values:
[AmplitudeValues2.ods](https://github.com/iliasam/Laser_tape_reverse_engineering/files/7541600/AmplitudeValues2.ods)

iliasam, 2021-11-16:
Thank you for data.
It looks really different to mine measurements, so it would require some time to process this data.

iliasam, 2021-11-18:
Could you do additional test:
Fully open receiver lens.
Place object to a "zero distance" position.
Confirm that "Phase1" is zero, and amplitude is near 1500 (as you measured earlier).
Try to close manually **laser** with something thin (knife is suitable). Check that light is not reflected to receiver lens.
Confirm that phase value is really begin to grow fast and reach 45 deg when amplitude is changing from 500 to 50 points.

maillemaker, 2021-11-27:
Thanks for working on this.  I have purchased some laser tape measures:

https://www.aliexpress.com/item/1005001772064136.html

And they came with 85A modules.  I would like to integrate a laser rangefinder with my Arduino.

MickTheMechanic, 2021-12-04:
Unfortunately I have damaged my remaining module. I must order some replacements, this means the project is temporarily on hold.

iliasam, 2021-12-04:
I'm orded one devise here: https://www.aliexpress.com/item/1005001772064136.html
I would try to investigate it.

bdevijve, 2021-12-08:
Hello,
I'm following this very interesting project and I own 2x AtuMan LS-1 / LS-1P from Banggood which run 85A as well...

MickTheMechanic, 2021-12-08:
I have a request for the new firmware, to make the baudrate user selectable by setting a gpio pin high or low.

For example is the pin is grounded then the device runs a lower baudrate, something like 56000, which is more Arduino friendly.

If the pins are floating/High then the device runs a higher baudrate, for using with PC utility.

Currently PA11 and PA12 are unused, and could be set low or high by modifying the button circuitboard included in the range finder.

Alternatively the pin TIM3/Ch2 (Buzzer pin) is also available, and it looks like it can also work as an input pin, which has the advantage of solder pads being located on the device board.

Is this something you would consider with the new release for 85a board?

iliasam, 2021-12-08:
@MickTheMechanic, please open new issue for baudrate request.

I'm still waiting for my new device with probably 85a module.

iliasam, 2021-12-18:
I have received my D30 laser rangefinder. It have 85A module.
I had notice that Li-ion battery is directly connected to the module. Battery voltage is usually in range 3.6-4.2V.
You can see two same ICs (with 5 legs) near power pads. Mine are marked "LB33", which can be decoded as MIC5205-3.3BM5.
Pinout is looking suitable. So these are Linear Voltage Regulator ICs, they produce 3.3V for the module.
That means that **input voltage must be higher than 3.3V.**
Also I tested how input voltage is displayed at the screen. All voltages below 3.7V are showed with low battery sign which is correct for Li-ion battery.

All previous modules were designed to be powered from two AAA batteries, so they have step-up DC-DC converter IC, which convert 2.5-3V from batteries to 3.5V. 85B module also have such module, so I'm not sure 85B module can be powered from Li-ion battery. Mine laser tape measure with 85B was powered by two AAA batteries.
Interesting thing: D35 Laser Tape Measure is designed to be powered from AA batteries so it can contain 85B module (of course, I'm not sure in it).

That is why I powered my 85A module with 3.7V.
I have replaced original firmware of 85A with mine own (for 85B module) with bootloader (without using programmer).
After that I have checked how firmware is working - and it is working OK without any modifications.
I tried to change signal amplitude by closing laser beam - it is not affecting the measured values.

bdevijve, 2021-12-18:
Very good point if the 85B firmware does not need any change for 85A then !
Can you point me to the flashing instructions as you performed ? (With bootloader) and last question : do you think we can power this device with 5V ?

iliasam, 2021-12-18:
See https://github.com/iliasam/Laser_tape_reverse_engineering/wiki/U85-module-programming.  
See also vis video, but I recommend you to use "Firmware_dist_calculation_fast_boot.hex".   
  
https://user-images.githubusercontent.com/10147086/146648515-e8e14f07-e56d-4fbf-a08f-5b903e57361a.mp4  

I'm not sure that it safe to power module with  5V.

MickTheMechanic, 2021-12-18:
Thankyou for doing these tests. It is possible that my input voltage was causing problems.

MickTheMechanic, 2021-12-19:
When you use the pc utility to measure distance, are you getting plausible measurements?

iliasam, 2021-12-19:
Yes, I compare distance measurements with my Bosh laser tape measure.

iliasam, 2021-12-20:
I have not tested my module with Arduino for a long time. 
Arduino sketch is published here: https://www.hackster.io/iliasam/making-a-cheap-laser-rangefinder-for-arduino-4dd849
256000 baudrate was used there.
I can't remember big problems there.
Notice, that  data speed is not very high - it is near 42*65 = 2730 bytes/s.
Yon can create nee issue au Github to report about baudrate problems.

MickTheMechanic, 2021-12-20:
I have very good results except when the object is between 180cm and 200cm from the laser, then I have very erratic and unlausible values. Can you please confirm if you experience similar problems with your module? Even with the incresed voltage I am still have problems..

MickTheMechanic, 2021-12-21:
I must apologise, it seems my white wallpaper has been causing the strange issues. I changed to a matt black surface and I am having good results

bdevijve, 2021-12-21:
Hi all, I took some measures yesterday : notice VBAT is between 3.7 and 4.2V : I will probably power it from 5V here;
But PWRON is between 2.7 et 3.0V, so I will connect a GPIO from the arduino / ESP32 to put enable this device;
I will probably connect a tasmota ESP32 to it; and upload the distance to MQTT...

( I notice UART speed settings on Tasmota is between 1200 and 115200... So i will probably have to change the U85 firmware to 115200 )

MickTheMechanic, 2021-12-25:
I have done a lot of testing and I continue to have problems with this module. I have also tested a new one and I have the same results. 

At certain distances the output becomes very erratic and implausible. If I use a Matt black surface then the problem is a bit better.

At first I thought it might be data corruption on the serial bus, but all other data is fine.

@iliasam do you experience similar problems with your module?

iliasam, 2021-12-25:
I do several tests with my 85A module and found that results could be really unstable when amplitude is more than 700-800 points.
I would do more tests and I will try to improve code.

maillemaker, 2021-12-29:
> I'm orded one devise here: https://www.aliexpress.com/item/1005001772064136.html I would try to investigate it.

I ordered this one to start.  It came with an 85A.

I have just ordered a second, different module:

https://www.aliexpress.com/item/1005001394260306.html

And it also came with an 85A.

maillemaker, 2021-12-29:
Hi all,

I am following this topic with great interest.  I am a pretty new Arduino user.  It's getting hard to follow this conversation across the various sites.  

As I understand things right now

- The 85A will work, but requires an input voltage greater than 3.3V.
- The 85A can be programmed via ST-Link or UART

I have an ST-Link.

Is the pinout for the 85A the same as the 85B?

85B from tutorial: https://github.com/iliasam/Laser_tape_reverse_engineering/raw/master/PCB_photos/module_85b/pinout.jpg

My 85A: https://i.imgur.com/Z0OGLr6.jpg

maillemaker, 2021-12-29:
It would be fantastic if this step-by-step article with pictures:

https://www.hackster.io/iliasam/making-a-cheap-laser-rangefinder-for-arduino-4dd849

Could be duplicated only using the 85A module.

MickTheMechanic, 2021-12-29:
@maillemaker Here is the guide for how to program the module. 

https://github.com/iliasam/Laser_tape_reverse_engineering/wiki/U85-module-programming

85a can be programmed exactly the same way as the 85b module, with the same firmware. It can be done either from STLink or via serial, both methods are documented well. You can use an Arduino board as the serial connection if you don’t have a usb to serial Converter, as pins RX and TX on the Arduino board are connected directly to the inbuilt USB to Serial converter on Arduino boards.

There is currently a bug that causes erratic values when the laser intensity is too high, so until @iliasam releases new firmware the current workaround is to use the color black as your reflective surface.

iliasam, 2021-12-29:
@MickTheMechanic Thanks for comment.
@maillemaker Currently I don't recommend to erase original firmware - because my firmware is working not stable at certain conditions, sorry for that. I'm afraid that this problem exists with 85b module too - I need time to confirm that.

The problem is that schematic of 85x modules differ somehow from previous schematic of 512/701/703 modules.
Measured phase value was always sensitive to amplitude, but newer modules have much more complicated dependence.
I don't know how it is possible, but I can see that phase dependence of amplitude is depending on "true" value of phase.
So at "true" 0 deg. measured phase is not depending on amplitude, but at "true" 90 deg measured phase is super sensitive to amplitude - phase is depending on amplitude and phase (sounds strange). So it is hard to measure phase,. 
I found this more than 4 month ago, and I implemented special compensation method than can handle with that, but you can see that this method is not working OK at certain conditions.  

All these investigations are need a lot of time, but I don't have a lot of free time, so research how to eliminate this problem could take a lot of time.  

Changing firmware with St-Link is more complicated, because you need to enable hardware reset is settings of programmer, and you programmer must have special reset pin.
 STM32CubeProgrammer can be used with St-Link, I have tested it.

[28 more replies]

closed 2022-11-14 by MickTheMechanic