iliasam/Laser_tape_reverse_engineering #7: Request firmware with lower baudrate

MickTheMechanic, 2021-12-21:
I would like to request a copy of firmware Firmware_dist_calculation_fast for U85 module, but with a lower baudrate of 115200. 

The reason for this is to eliminate data corruption due to high baudrate as a reason for the unplausible values mentioned in #5

bdevijve, 2021-12-21:
Hi, +1 as 115200 is the max speed that can be used on Tasmota...

bdevijve, 2021-12-21:
As I plan to connect this device to Tasmota and upload the distance to MQTT, I will need to lower the UART speed to 115200 at max; but probably to lower the measurement frequency to let's say 1Hz... I will try to make the change on the provided firmware on my own and tell you about the result...

iliasam, 2021-12-21:
I have published FW for 115200 baudrate. I have not tested it, but I think that it should work. https://github.com/iliasam/Laser_tape_reverse_engineering/blob/master/Code/CortexM0/Firmware_dist_calculation_fast_boot_115200.hex

iliasam, 2021-12-25:
Did someone tested new firmware?

MickTheMechanic, 2021-12-25:
I hope to test on Sunday, I have been very busy the last days.

MickTheMechanic, 2022-11-14:
Firmware is working well, thanks!

closed 2022-11-14 by MickTheMechanic