iliasam/Laser_tape_reverse_engineering #8: 88B planned ?

skiphansen, 2022-07-25:
Great project!  

Unfortunately the X-40's are very scarce  now.  I bought the only one I could find on ebay and it just arrived ... with an 88B module.  Any chance you're going to support it someday?  Any suggestions where I might find a X-40 with the right module?

Thanks for the amazing work in any case ... wish I could give you 4 stars!

Skip

skiphansen, 2022-07-26:
I looks like the 88B module already has serial which is what I need.  But I can't find any documentation on the serial protocol, sigh.

Here's some info I mined from https://www.alibaba.com/product-detail/Phase-Shift-M88B-Laser-Distance-Measurement_60696810434.html
[image]
[image]

iliasam, 2022-07-26:
Hello.
Unfortunately, support for M88 modules is not planned - I don't have such module.
Newer U85x modules are supported: [Link](https://github.com/iliasam/Laser_tape_reverse_engineering/wiki/U85-laser-rangefinder-modules)
Notice that firmware for cheap laser tape measure modules is differ from more expensive "UART" versions of that modules - UART is disabled in laser tape measures.
Looks like that this is manual for M88: [Link](https://forum.arduino.cc/uploads/short-url/hZh63mc9cPaRPq44QVSyfyYtYYF.pdf)

skiphansen, 2022-07-26:
Thanks for the quick response! I wish shipping from China was as fast!
I'll order some U85x versions and revisit my project in a month or so.

closed 2023-06-24 by iliasam