iliasam/Laser_tape_reverse_engineering #9: Voltage regulation 85B Module

MickTheMechanic, 2022-11-17:
I am using multiple 85B modules for a project, running your firmware. 

I see you have inspected the boards thourougly, what is the highest voltage you would reccomend to power such modules?

iliasam, 2022-11-18:
They have internal DC-DC step up converter, which is generating 3.5V.
So I think that max input voltage can be near 3.3V.
But I recommend to use 3V.

MickTheMechanic, 2022-11-18:
Do you have any info on the voltage regulation IC? Maybe you found a part number?

iliasam, 2022-11-23:
It is marked as "FB3MKE"
https://github.com/iliasam/Laser_tape_reverse_engineering/issues/5#issuecomment-962657155
This can be FP6714: https://datasheet.lcsc.com/lcsc/1912111437_Fitipower-Integrated-Tech-FP6714S6CTR_C389170.pdf

jian-li, 2023-05-02:
Hello, we have a higher-frequency measurement requirement, Can I have your personal communication like wechat or facebook to get contacted?

iliasam, 2023-05-02:
Hello. You can write to: iliasam3@gmail.com

closed 2023-06-24 by iliasam