iliasam/Laser_tape_reverse_engineering #10: Arduino Nano Not working

manisaranbala, 2023-11-07:
My device U85B .Nano RX TX connect not reading.but TTL good reading .we need Arduino Nano or 2560 program connect Desktop.it is possible?

iliasam, 2023-11-07:
Sorry, I don't understand your question.
Can you describe step by step what you already tried?
What hardware do you have?

manisaranbala, 2023-11-07:
[image]
my device 40m 85B TTL MODE Reading good. but not reading Arduino Nano  serial communication

closed 2023-11-07 by manisaranbala

manisaranbala, 2023-11-14:
https://github.com/iliasam/Laser_tape_reverse_engineering/assets/5657183/c912efa5-3983-4914-af42-9ec90e0fe0e4

laser light power is down in few minutes. and also the reading is differ so how can solve it

manisaranbala, 2023-11-14:
set volt 2.2 and 3 and 3.2 same Problems

manisaranbala, 2023-11-14:
https://github.com/iliasam/Laser_tape_reverse_engineering/assets/5657183/d4dd4189-3ae5-41b0-8331-4e6f3562143d

manisaranbala, 2023-11-14:
[Result_2023-11-14 12-49-20.csv](https://github.com/iliasam/Laser_tape_reverse_engineering/files/13345703/Result_2023-11-14.12-49-20.csv)

iliasam, 2023-11-14:
Do you have this problem with USB-UART converter?
Also, try to disconnect RX line of U85B module - maybe 5V from Arduino is bad for U85B.

manisaranbala, 2023-11-14:
internal battery (AAA)same problem. Need continue Data Reading possible?

manisaranbala, 2023-11-14:
[image]
[image]
[image]
![WhatsApp Image 2023-11-14 at 4 13 27 PM (1)
[image]
](https://github.com/iliasam/Laser_tape_reverse_engineering/assets/5657183/5a7830c9-3afb-4511-bf8e-cda51232bbba)
How to find  this model good or bad

manisaranbala, 2023-11-14:
[image]

iliasam, 2023-11-14:
Again:
"Do you have this problem with USB-UART converter?
Also, try to disconnect RX line of U85B module - maybe 5V from Arduino is bad for U85B"

manisaranbala, 2023-11-14:
https://github.com/iliasam/Laser_tape_reverse_engineering/assets/5657183/57c5d04f-e0e8-43cd-845b-881dcae52d1a

manisaranbala, 2023-11-14:
Amplitude Reduse Automatically . That time  mismatched calculation . how to solve this problems?