iliasam/Laser_tape_reverse_engineering #11: Request for new firmware, 0.1mm resolution and max distance of ~3 meters

MickTheMechanic, 2023-11-10:
Request for a new firmware with higher resolution and shorter maximum distance. 

I would like to increase the resloution to 0.1mm and change the frequencies to optimize for a short distance of around 3 meters.

Is this something you would be interested in working together on?

iliasam, 2023-11-10:
I'm not sure that it is physically possible.

MickTheMechanic, 2023-11-10:
I am talking about resolution, as in the output format, xxx.x instead of xxx
And regarding the frequencies, you mentioned the frequencies you chose will only work up to ~37m, can we choose other frequencies that are better suited to a shorter range, and sacrifice distance for accuracy?

MickTheMechanic, 2023-11-10:
You spoke about it here: #4 
And in your blog;

> If the modulation frequency is 10 MHz, then the measured distance can reach up to 15 meters, and when the distance changes from 0 to 15 meters, the phase difference will change from 0 to 360 degrees. A change in the phase shift by 1 degree in this case corresponds to a movement of the object by approximately 4 cm.
....
The simplest case is to use two frequencies, at a low frequency they approximately determine the distance to an object (but the maximum distance is still limited), at a high frequency they determine the distance with the required accuracy - with the same accuracy of measuring the phase shift, when using a high frequency the accuracy of measuring the distance will be noticeably higher. <
Is there any benifit to modifying the frequencies for a shorter distance?

And is there any point in the current distrance calculation where we can round differently or multiply by 10 or something similar to get it to output in 0.1mm increments instead of mm inceraments?

MickTheMechanic, 2023-11-11:
So if i understand correctly, 50MHz has a wavelength of ca. 6 meters, so it should be suitable for measuring a distance of 3 meters.
That would give us ca 83mm or 0.83 cm per 1 degree of phase shift.
The we can use a much higher frequency than before as we only need to cover a distance of 0.83cm, which would give us higher accuracy.

Do i understand the working principle here? If so, is the hardware capable of this, and is the software easily modified in this regard?

iliasam, 2023-11-11:
"I am talking about resolution, as in the output format, xxx.x instead of xxx. (I understand the accuracy will still be in the mm range)"
Probably it is possible to change "WAVE_L1" and others to x10, but I do not sure that this will help change resolution.

To increase resolution, you need to increase modulation frequency, currently max is 193 MHz. More frequency value -> shorter wavelength, but phase measurement resolution is the same, so distance resolution increases. 
I can't remember, if I tried to increase them or not in my experiments. Of course, some HW limit exists.

MickTheMechanic, 2023-11-11:
Can I just change those values, or do I also need to modify the calculation functions?

iliasam, 2023-11-11:
Possibly yes, but I'm not 100% sure. I write that code a long ago.

MickTheMechanic, 2023-11-13:
I have been reading your blog posts, while the theme is very complicated for me, slowly I start to understand.

What do you think about selecting frequencies that are further apart, in order to reduce Dmax and also reduce measurement error?

iliasam, 2023-11-13:
Decreasing Dmax can reduce ambiguity errors (wrong calculated number of integer waves). This error is in 1-2m range.
As I said: 
"To increase resolution, you need to increase modulation frequency, currently max is 193 MHz. More frequency value -> shorter wavelength, but phase measurement resolution is the same, so distance resolution increases."

MickTheMechanic, 2023-11-14:
Ok, i have read through the code and it seems to me like there are two places I need to make changes, I need to change the frequencies that the pll generates in pll_functions.c and I need to change the wavelengths in distance_calc.c

Can you provide me with information about how to control the pll? Your code  was not very clear in this regard, can you explain to me how it works?

`

//162.5 + 162.505
void pll_set_frequency_1(void)
{
  pll_change_freq(26, 0, 1, 1250);//162.5 + 162.505
}

//191.5 + 191.505
void pll_set_frequency_2(void)
{
  pll_change_freq(30, 800, 801, 1250);//191.5 + 191.505
}

//193.5 + 193.505
void pll_set_frequency_3(void)
{
  pll_change_freq(30, 1200, 1201, 1250);//193.5 + 193.505
}

`

iliasam, 2023-11-14:
I added small description:

> //pll_mult - common multiplication
//plla_coef/pllb_coef - outputA/outputB coefficients
//pll_div - common divider
//freqA = 25mhz*(pll_mult + plla_coef/pll_div) / 4
//freqB = 25mhz*(pll_mult + pllb_coef/pll_div) / 4
//Divider to 4 can be switched
void pll_change_freq(uint32_t pll_mult, uint32_t plla_coef, uint32_t pllb_coef, uint32_t pll_div)

Frequencies are taken from original FW. Not all combinations are working well.
  plla_coef/pllb_coef must differ to "1" to keep difference between frequencies = 5kHz.
You can try to change "pll_mult" values, but you will need to change "WAVE_L1" and others.

MickTheMechanic, 2024-02-07:
I have some questions about the this device. I see that the frequencies used to measure the phase shift are in the mHz range, and if I understand correctly the Goertzel algorithm is used to process the frequencies. 

I have experience with the Goertzel algorithm from other projects, my experience tells me that the maximum frequency the Goertzel algorithm can detect is half the sample frequency of the ADC. 

How do you get around this? Is the ADC really so fast on this device?

iliasam, 2024-02-07:
I can advise you to look at the device structure schematic at the repository description and read translated article.
You can see that PLL is generating two high frequencies, but with 5kHz difference - Freq A and Freq B. They are mixed with each other at two places - at the special mixer and at the APD. After additional LPF filtering you have two signals with 5 kHz frequency. Those signals are processed by MCU.

MickTheMechanic, 2024-04-13:
I am trying to modify the firmware, but the STMCubeIDE doesnt recognise your source code/mcu configurations. 

Which compiler/IDE did you use?

iliasam, 2024-04-13:
https://github.com/iliasam/Laser_tape_reverse_engineering/blob/master/Code/Readme.txt
"Used IDE - IAR 7.50"

MickTheMechanic, 2024-04-13:
Thanks.

I plan to try something different. I want to implement a mode that modulates the laser at a lower frequency for another purpose, and be able to switch back to range finder mode.

Let’s say I want to modulate the laser at say 700hz so that an external device can detect its presence via a photodiode that samples and uses Goertzel algorithm to detect that frequency.

What would be the best way to modulate the laser?

In your opinion should I reconfigure the pll to inject 700hz frequency via LASER_Signal, or should I modify the PWM frequency for the laser voltage?

iliasam, 2024-04-13:
I tried to modulate laser current DAC line (line10) at 1 kHz, but I get bad results (sorry, but I don't remember what was wrong with them).
Maybe U85 modules are working different.
I don't that that C18/C23 capacitors will pass such low frequency after PLL.
Also, this is off topic for this issue.

MickTheMechanic, 2024-04-14:
How did you get it? I cant even find a way to download it as a private individual.

iliasam, 2024-04-14:
Sorry, I don't understand your question.

If this issue is completed, please close it. You can write to my e-mail (see it in my profile) or create new issue.

MickTheMechanic, 2024-04-14:
I sent you an email

closed 2024-04-14 by MickTheMechanic