iliasam/Laser_tape_reverse_engineering #15: Question on source code

epsi1on, 2024-07-11:
I was looking through source code, in line 10 of file [distance_calc.c](https://github.com/iliasam/Laser_tape_reverse_engineering/blob/c0a90e39ef6ebc42945bdd2fe24ddc16bc5c0b8b/Code/Firmware_dist_calculation_fast/src/distance_calc.c#L10) it seems that there are three frequencies used:

```
#define WAVE_L1         59036 //wavelength for 162.5Mhz in mm * 32
#define WAVE_L2         50095 //wavelength for 191.5Mhz in mm * 32
#define WAVE_L3         49578 //wavelength for 193.5Mhz in mm * 32
```

which is `162.5`, `191.5` and `193.5`. Could you please give information about whether these are modulated frequencies? if yes, do you advice for me to edit the code to use multiplies of 2?

For example `193.5` and `193.5/2=96MHz` and so on until it gets to say 1 MHZ. then one would be able to tell the distance more precisely and possibly without [brute force](https://github.com/iliasam/Laser_tape_reverse_engineering/blob/c0a90e39ef6ebc42945bdd2fe24ddc16bc5c0b8b/Code/Firmware_dist_calculation_fast/src/distance_calc.c#L67) algorithm.
Thanks for sharing great experience...

epsi1on, 2024-07-11:
Assuming we know that the distance is less than 150m far, one would first try with `1Mhz` modulation, with wavelength of `300m`.  then from phase diff, one will understand distance is about 70 meter. retry with wavelength of 50 and will turn out phase diff is about to 20.5m. and so on until it gets to the maximum frequency which result best resolution. this way the maximum range could extend to more that 37.5m (currently).
Thanks

iliasam, 2024-07-11:
Sorry, I don't understand your questions well.
Have you read this article: https://habr.com/ru/articles/327642/ ?

162.5 Mhz, 191.5 Mhz, 193.5 Mhz are modulation frequencies. They are set here: https://github.com/iliasam/Laser_tape_reverse_engineering/blob/master/Code/Firmware_dist_calculation_fast/src/pll_functions.c
It is not possible to set frequency less that a certain limit  - it is limited by hardware.
I did some experiments earlier, 25Mhz gave too small signal.

Also, 40m limit is based on hardware sensitivity - original laser tape measure was measuring such distance slowly, because signal is very weak. I don't think that it is possible to make measurements better.

epsi1on, 2024-07-11:
Thanks for information.
> It is not possible to set frequency less that a certain limit - it is limited by hardware. I did some experiments earlier, 25Mhz gave too small signal.
Could you please give more information about what are the minimum and maximum possible modulate frequencies which hardware allow me to set? what you mean by 25MHz gave small signal?

> Also, 40m limit is based on hardware sensitivity - original laser tape measure was measuring such distance slowly, because signal is very weak. I don't think that it is possible to make measurements better.

I was thinking to use a corner prism to reflect back the laser beam, this way the returned laser will not get weak so hopefully i could be able to measure larger distances. say 200m with a quality hand made (!) corner reflector.

Thanks again

iliasam, 2024-07-11:
"what are the minimum and maximum possible modulate frequencies "
I don't remember them well, but I know that they are dependent on hardware type: https://github.com/iliasam/Laser_tape_reverse_engineering/wiki/Known-modules
Maximum limit is Si5351 limit - near 250MHz, if I not mistaken.

"what you mean by 25MHz gave small signal?"
Signal amplitude is 2-5 times smaller when 25MHz modulation is used, comparing to "traditional" 150-200MHz modulation frequency.

If you are planning to use corner prism, you can try to make L3 frequency 193.5MHz more closer to L2, 192.0Mhz for example.
Yo will need to change "WAVE_L3" value and "pll_set_frequency_3()" implementation.

Notice that it can be problematic to find laser tape measure hardware supported by this project now.

epsi1on, 2024-07-11:
> Notice that it can be problematic to find laser tape measure hardware supported by this project now.

Actually i'm trying to find one.

epsi1on, 2024-07-11:
The modulation frequency is ~193MHZ, but how stm32's adc is used to find phase diff such relative high frequency signal?
the schematics it shows that a 5khz signal is passed to micro controller.

[image]

Could you please describe it for a newby electronic guy like me?
Thanks

epsi1on, 2024-07-11:
let me read the article carefully, then will ask question if any.
Thanks

iliasam, 2024-07-11:
It is all described here - https://habr.com/ru/articles/327642/
Use goggle translate if you need it.
See this image: https://habrastorage.org/r/w1560/getpro/habr/post_images/b38/81b/cc5/b3881bcc5a8babc7e9d23821da51061e.png
And read this: https://en.wikipedia.org/wiki/Heterodyne

epsi1on, 2024-07-12:
> It is all described here - https://habr.com/ru/articles/327642/ Use goggle translate if you need it. See this image: https://habrastorage.org/r/w1560/getpro/habr/post_images/b38/81b/cc5/b3881bcc5a8babc7e9d23821da51061e.png And read this: https://en.wikipedia.org/wiki/Heterodyne

I got all answers by reading the article and your post. the key for conversion from high frequency to low frequency was [Heterodyne](https://en.wikipedia.org/wiki/Heterodyne).
Thanks you for sharing knowledge...

closed 2024-07-12 by epsi1on