iliasam/Laser_tape_reverse_engineering #16: B2A module

epsi1on, 2024-07-13:
Hello,
I have a module marked as `B2A` as in image. in known modules also you marked a model named `MB2A` as not supported with this firmware. but the `B2A` module's visuals is very similar to `85A` module which is supported by the firmware.
Here are `B2A` images:

[image]

[image]

and here is `85A` and `85B` modules: ([link](https://github.com/iliasam/Laser_tape_reverse_engineering/wiki/U85-laser-rangefinder-modules))
The `B2A` have an extra chip regard to `85A` which i think it contains accelerometer (to show angle relative to gravity).

Do you think it is supported?
Thanks

iliasam, 2024-07-13:
You can see from the description: https://github.com/iliasam/Laser_tape_reverse_engineering/wiki/Known-modules
that B2A has STM32G03/04x MCU (I don't know real type).
U85 modules have STM32F03x MCU. 
So current firmware is need to be ported. 
Firmware for U85 is written with SPL, while there is no SPL for STM32G.

epsi1on, 2024-07-13:
Hello, and thanks for the reply.
While there is no SPL for the STM32G, do you suggest that HAL library be used for porting the firmware?
I'm very interested to port firmware.

iliasam, 2024-07-13:
Yes.

epsi1on, 2024-07-13:
Thanks for reply.
Can [this document](https://github.com/iliasam/Laser_tape_reverse_engineering/wiki/U85-module-programming) be used for programming the B2A module?

I'm using such device for programming (seller claimed that it is not chip fake one, he said it have ID which i'm not sure what it is.) 

[image]

iliasam, 2024-07-13:
Maybe yes.
I hope, you have seen this - " You programmer must have a special reset pin (cheap Chinese St-Link dongles don't have it)."
You programmer don't have it.

You can try to use bootloader.

epsi1on, 2024-07-14:
Thanks for the reply.
Can i create a backup of device firmware in order to be able to restore it?

Also do you suggest me to get one of these programmers

[image]

i'm interested to get one for this and for future usages, unless it doesn't make any help for me now.

epsi1on, 2024-07-17:
Hi,
I've got one of these programmers
[image]

First question is it intended that NRST pin on this programmer work?

I 've connected the module to programmer like this:
```
Connect "PWRON" line to VBAT line - to power on module.
You need to connect SWDIO, SWCLK, NRST, GND to your programmer (ST-Link for example).
Also it is recommended to connect VBAT to TVCC (target VCC) input pin if your programmer has it (like original St-Link).
Notice that NRST must be connected! You programmer must have a special reset pin (cheap Chinese St-Link dongles don't have it).
You need to select "Connect under Reset" and Reset mode: "Hardware Reset" in options of your programmer software.
This method is recommending if you are planning to change source code.
```

but i've got following error:

[image]

iliasam, 2024-07-17:
Configuration of the CubeProgrammer utility is looking OK.
Target Voltage = 2.9V is looking too small.
How do you connected power from module to programmer? (How pin1 of the programmer is connected?)
How you are powering your module?
What is the voltage at si5351 (its pin1 is VDD)?

epsi1on, 2024-07-17:
Thanks for the reply.
I've did another test.

[image]

Here is how things are connected:

| B2A     | STLINK      |
|---------|-------------|
| 1-SWDIO | 7-SWDIO     |
| 2-SWCLK | 9-SWCLK     |
| 3-RX    | Not Connect |
| 4-NRST  | 15-RESET    |
| 5-BOOT0 | Not Connect |
| 6-GND   | 4- GND      |
| 7-PWRON | 1- VTREF    |
| 8-Vbat  | 1-VTREF     |

And here is what i do assume pinout of my stlink (according to [utmel.com](https://www.utmel.com/components/stlinkv2-in-circuit-debugger-pinout-datasheet-and-programming?id=237))

[image]

The voltage at pin1 (VDD) of si5351 is 3.26V 

Thanks

epsi1on, 2024-07-17:
Another odd thing is the laser diode on module is constantly ON after i did module to STlink
Thanks

epsi1on, 2024-07-18:
Hi,
Thanks for reply.

1
--
> "How you are powering your module?" 

Not sure, two pins on B2A module (7-PWRON and 8-Vbat) are connected to VTRef an nowhere else. no other batteries etc. 
[image]

2
--
> What is Vbat value? 

You mean the voltage of `VBat` pin on module?

> What was the original battery of your module?

[image]

`A*uman` (the `t` is replaced by `*`)
`651720 3.7v`
`0.82Wh 2402`

[image]
[image]

3
--
> Check this point -
> If it is 3.3V, so this is MCU VDD, you can try to use is as source for programmer VTref.
Multi meter said it is 3.25v, so i think it is 3.3v. but could you please give me some info on how can I connect it into VTref?

Thanks

iliasam, 2024-07-18:
VTref of the st-link programmer is INPUT! it is power for internal programmer voltage-level converters.
So you need to have external power source with V=3.7-4.2V. Maybe 3.3V will be suitable, but I'm not sure. It must be connected to Vbat and GND of the b2a module. 

PWRON of b2a also must be connected to that power source.
VTref is better to be connected to MCU VDD with 3.3V.

Looks like that now your module is powered by voltage leakage through data pins.

*******
You can try co connect pins 1 and pin 19 of your programmer, and do not connect anything other to that pins, so programmer will have 3.3V for its internal voltage-level converters.
https://i.sstatic.net/rIMNJ.jpg

epsi1on, 2024-07-18:
Thanks for information, and sorry for bothering you.

i did some changes, now a 3.7v battery is also connected like this:

| B2A     | STLINK      |
|---------|-------------|
| 1-SWDIO | 7-SWDIO     |
| 2-SWCLK | 9-SWCLK     |
| 3-RX    | Not Connect |
| 4-NRST  | 15-RESET    |
| 5-BOOT0 | Not Connect |
| 6-GND   | 4- GND      |
|MCU VDD (solder new wire to the cap pin) | Vtref|

| B2A     | Battery     |
|---------|-------------|
| 6-GND   | GND |
| 7-PWRON | +3.7v  |
| 8-VBAT | +3.7v    |

Voltage at VBATT and PWRON are 3.66v
voltage at MCU VDD pin is 3.26v
a new wire from vdd is added to vtref pin.

but still get message:
[image]

iliasam, 2024-07-18:
"voltage at MCU VDD (and VTref pin on STLINK) is 3.26v"
But utility is writing that voltage is 2.99-3.01V. This is strange for me.

epsi1on, 2024-07-19:
Did a retry with boot0 and UART method, got this error:
Can it be some sort of reverse engineering protection?

[image]

iliasam, 2024-07-19:
I don't think so. Your MCU is silent.
Looks like STM made changes at STM32G bootloader activation: https://community.st.com/t5/stm32-mcus-products/has-anyone-gotten-the-boot0-pin-to-work-on-an-stm32g071-solved/m-p/338772
https://electronics.stackexchange.com/questions/598336/stm32g030-boot0-pin-swclk
Maybe connecting SWCLK to 3.3V can help, maybe not.

Is MCU of your module still working without connections, when LCD with keys connected?

epsi1on, 2024-07-21:
hi
> Is MCU of your module still working without connections, when LCD with keys connected?

Yes, when i remove programmer and connect battery and keypad to the module, it works just fine as before.
Let me try to connect SWCLK to 3v3 and retry...
Thanks for tips, i really appreciate it...
Have a nice day

epsi1on, 2024-07-21:
just another question which is not related to the µc, would be thankful if you clear me.

in this [file](https://github.com/iliasam/Laser_tape_reverse_engineering/blob/master/Schematic/module_701A/schematic_PLL_and_laser.png), at the Laser Diode D3, how the sin wave is applied to the D3 anode?
I think voltage at D3 anode is sin wave, but what is the min and max in volt scale? i think it should have offset from zero but not sure.

Thanks again

iliasam, 2024-07-21:
" but what is the min and max in volt scale?"
Sorry, but I can't say - I don't have a suitable oscilloscope when I was doing experiments with that module.
Laser diode circuit is generating voltage offset - it is near 2V.

epsi1on, 2024-07-28:
Lets close this issue for now.
Thanks for all information.

closed 2024-07-28 by epsi1on

epsi1on, 2024-09-25:
> All my firmware is based on SPL libraries, while STM32G has HAL only. So it will take a lot of time for research and porting code.

Sorry, you did already note that, and i almost forgot about it.

epsi1on, 2024-10-02:
What if i do the port? you only give me instruction on how to reprogram the module. Already tried to connect module to programmer, but no success as discussed in this issue...
Thanks

iliasam, 2024-10-02:
"What if i do the port?" How do you plan porting FW without working hardware?

epsi1on, 2024-10-02:
> "What if i do the port?" How do you plan porting FW without working hardware?

Without working hardware, it is not possible.
but if you consider buying a b2a module (which i'll pay for it, which is like $25), there is a higher chance that you can find a way to reprogram the module. like pin change etc which i do not know. then you give me the instruction on how to connect module to programmer and reprogram it and this way i'll have a working hardware which i can work on...
Thanks

iliasam, 2024-10-02:
OK, I will think about it.

iliasam, 2024-10-14:
Please, can you give me a marking of PLL - it is IC with 10 pins?

epsi1on, 2024-10-14:
> Please, can you give me a marking of PLL - it is IC with 10 pins?

Hi, thank you for the reply.
I actually was not able to see the marking on the PLL chip (they are small and i do not have magnifier/microscope at time) but it is 10 pin.
I did attached a photo of my B2A module in first post of this issue (took by macro lens of phone).
Thanks again.

epsi1on, 2024-10-14:
Someone else try to find out the mcu type of B2A module, but not success: 

https://www.eevblog.com/forum/microcontrollers/identify-mcu/

seems the PLL marking for B2A module of that person is:
6M5351A
2316HT

buy mine is not clearly visible.

iliasam, 2024-10-14:
I finally bought laser tape measure for myself.
MCU (NM002) is probably STM32G041K8.  
I was able to connect to in using both original ST-LINK and cheap Chinese St-Link clone. Looks like that NRST is not needed.  
Firmware is locked, as expected. It can be erased easily with St-Link (see description at Hackster.io, not Wiki).  
You can see pinout here: https://github.com/iliasam/Laser_tape_reverse_engineering/blob/add_B2A/PCB_photos/module_B2A/B2A_reverse.JPG
But it looks like that they really changed PLL - it is GRANDMICRO GM5351A.
And it looks kike that it is not protocol compatible with older Si5351.
And I can't find GM5351A register map ((

epsi1on, 2024-10-15:
Thank you for the information,
could you please give me a crypto wallet address for donation?

epsi1on, 2024-10-15:
I think you could make an anonymous offline BTC paperwallet within minutes and simply, with www.bitaddress.org. print it for yourself and keep the private key secret.
And do not touch it like for decades :)  hopefully your BTC will remain safe on the blockchain.

iliasam, 2024-10-23:
I had published code for B2A: https://github.com/iliasam/Laser_tape_reverse_engineering/wiki/B2A-laser-rangefinder-modules
You can try it.
You can see my wallet at https://github.com/iliasam page.